citruz/beacontools #6: iBeacon battery charge level

maltitco, 2017-10-05:
It is possible to check on beacon battery charge level?

citruz, 2017-10-05:
As far a I know, the battery level is not part of the standard iBeacon specification and only available when you use the vendor's proprietary protocols. Therefore the library does not support it. 
Which beacons are you using?

maltitco, 2017-10-05:
I use IBc41 [https://www.dfrobot.com/index.php?route=product/product&product_id=1296&search=tel0095&description=true#.VhIzRSuG2Ht](url)

citruz, 2017-10-05:
Ok yes, it is as I said. The battery charge level is only visible using the configuration app provided by the vendor. It is not part of the iBeacon advertisement. The information is probably contained in BLE packets which are encoded using a vendor-specific format. You could try to sniff these packets and reverse-engineer the format. But I cannot help you with that as I don't have one of these beacons.

maltitco, 2017-10-05:
@citruz Thank you for the information.

closed 2017-10-05 by maltitco