citruz/beacontools #14: ConstructError("subcon[N] syntax can only be used for arrays use GreedyRange(subcon) instead?"

zSheikhAli, 2018-04-04:
I have been trying to run my rapberry Pi 3 with the beacontools library 
but i keep getting the error 
ConstructError("subcon[N] syntax can only be used for arrays use GreedyRange(subcon) instead?"

I am running it using pyhton2.7
Any idea how i can resolve this?

citruz, 2018-04-09:
Sorry for that. The developers of the the construct library published a change in version 2.9.41 which broke beacontools. Unfortunately, they don't do semantic versioning.
I just released version 1.2.3 which fixes this issue. Update using `pip install beacontools --upgrade`.

closed 2018-04-09 by citruz

zSheikhAli, 2018-04-12:
Thanks for your reply 
Unfortunately I updated it and I got the same error again while running the iBeacon scanning example uploaded 

Could it be possible that we’re testing the scanning in a wrong way?

citruz, 2018-04-15:
Sorry I can't reproduce the error. The `scanner_ibeacon_example.py` sample runs fine. Are you sure you upgraded the library? If yes, can you post the complete error message and stacktrace?

zSheikhAli, 2018-04-16:
the error I posted is resolved, thank you
I have a question though, how can I modify the scanner_ibeacon_example.py to scan for packets with specific ID and display the scanned packet information in the following format
uuid  major  minor power

citruz, 2018-04-16:
You can just extend the constructor of the `IBeaconFilter` like this:
```
IBeaconFilter(uuid="e5b9e3a6-27e2-4c36-a257-7698da5fc140", minor=1234)
```
Of course, scanning only by minor is possible as well:
```
IBeaconFilter(minor=1234)
```

zSheikhAli, 2018-04-17:
hello Again :)
I tried the scanning for a specific minor, and I am trying to read 18 packets 
I was hoping for a better delay than the scanning using a bash script but the results were even worst and I got a longer delay using the beacontools 
what could be the reason?