citruz/beacontools #16: need help!!

jesicanewton, 2018-04-19:
Hello 
thank you a lot for this great code 
You are a life saver!
However, I need some help regarding the scanning algorithm 
how can I edit the scanning so that I only hear from one specific mac address 
I do not want to scan for all BLE packets in air, I just need to lock the scanning to one mac address 
Please help me
Thank you 
Jesy

jesicanewton, 2018-04-19:
I did this today but the scanning is still slow
What I want is not to read all packets and then filter them based on some field because this slows down the process
I need to have the scanning initially locked to one device and only hear from it  

conorslater

jesicanewton, 2018-04-19:
filtr = self.bluez.hci_filter_new()
        self.bluez.hci_filter_all_events(filtr)
        self.bluez.hci_filter_set_ptype(filtr, self.bluez.HCI_EVENT_PKT)
        self.socket.setsockopt(self.bluez.SOL_HCI, self.bluez.HCI_FILTER, filtr)

Can you please explain to me what these four lines do in the Scanner.py code line 89-92 
Looking forward to your help!

conorslater, 2018-04-19:
The HCI filter filters for expected events or commands detailed in this table:
[http://www.lisha.ufsc.br/teaching/shi/ine5346-2003-1/work/bluetooth/hci_commands.html](url)

Unfortunately, it doesn't filter MAC addresses.

I did find that the `hcitool` has a whitelist functionality for MAC addresses which could be used as part of a workaround.

`hcitool lewladd 12:3b:6a:1b:4c:65`
`hcitool lescan --whitelist`

Although, I couldn't find this functionality implemented in pybluez.

jesicanewton, 2018-04-21:
Hello, I am trying to open the link but I am getting 404 
this is not the web page you are looking for 
Any idea why?

conorslater, 2018-04-21:
Sorry, I didn't enter the url correctly.

Try it now.

jesicanewton, 2018-04-22:
After adding my device to the whitelist how can I receive packets from it 
I realized the the lescan only scans for devices in the whitelist but not their packets !

conorslater, 2018-04-22:
@jesicanewton you can use `hcidump --raw` to output the packets after you run `hcitool lescan`.
e.g.
`hcitool lewladd 12:3b:6a:1b:4c:65`
`hcitool lescan --whitelist > /dev/null &`
`hcidump --raw`

However, I discovered a way to allow beacontools to take advantage of the whitelisting functionality of `hcitool`

Run this bash script `whitelist_MAC.sh`...
```
#!/usr/bin/env bash

# restart hci
hciconfig hci0 down
hciconfig hci0 up

if [ $1 = "clear" ]; then
    echo Clearing whitelist...
    hcitool lewlclr
else

    # add MAC addresses to whitelist
    for mac in "$@"
    do
        echo Whitelisting $mac....
        hcitool lewladd "$mac"
    done

    # start scanning
    hcitool lescan --whitelist > /dev/null &
    hcitool_pid=$!

    # kill process
    pkill $hcitool_pid

fi

echo done

exit 0
```

...with the mac addresses you want to whitelist `./whitelist_MAC.sh 12:3b:6a:1b:4c:65 12:3b:6a:1b:4c:67` and then run your beacontools program (e.g. `python3 scanner_ibeacon_example.py`)

You should only recieve packets from the MACs whitelisted.

You can clear the whitelist with `./whitelist_MAC.sh clear`

I'm not sure how it works and it is kind of a hack, but it appears that if you run `hcitool lescan --whitelist` and then kill the process, the whitelist is stored for beacontools to use.

Let me know if this works for you.

citruz, 2018-05-23:
Closing this because of inactivity.

closed 2018-05-23 by citruz