citruz/beacontools #17: scanner.start() results in `_bluetooth.error: (1, 'Operation not permitted')`

williamflynt, 2018-05-23:
First: I know this is a permissions issue - and I also see the step in the setup instructions to mitigate it.

## System:
Ubuntu 17.10 x64
Pycharm 2018.1.3 (developing here)
Python 3.6 in a virtualenv

## Steps to Replicate:
~~~
pip install beacontools
# install libbluetooth headers and libpcap2
sudo apt-get install python-dev libbluetooth-dev libcap2-bin
# grant the python executable permission to access raw socket data
sudo setcap 'cap_net_raw,cap_net_admin+eip' $(readlink -f $(which python))
# install beacontools with scanning support
pip install beacontools[scan]
~~~

Then copy-paste example code from *Scanner* subsection (`scanner = BeaconScanner...`)

## Traceback
~~~
Exception in thread Thread-5:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/home/william/PycharmProjects/bt-beacon/venv/lib/python3.6/site-packages/beacontools/scanner.py", line 94, in run
    self.toggle_scan(True)
  File "/home/william/PycharmProjects/bt-beacon/venv/lib/python3.6/site-packages/beacontools/scanner.py", line 110, in toggle_scan
    self.bluez.hci_send_cmd(self.socket, OGF_LE_CTL, OCF_LE_SET_SCAN_ENABLE, command)
_bluetooth.error: (1, 'Operation not permitted')
~~~

## Things I Tried
Various forms of `sudo setcap...` including `sudo setcap 'cap_net_raw,cap_net_admin+eip' $(which hcitool)'.

Has anyone else experienced and solved this issue?

citruz, 2018-05-23:
Try executing ```sudo setcap 'cap_net_raw,cap_net_admin+eip' $(readlink -f $(which python))```inside your venv.

williamflynt, 2018-05-24:
No luck. This is what I've tried before. I'm sure it's something on my end, but I can't for the life of me imagine why it wouldn't work on such a standard setup.
[image]

citruz, 2018-05-25:
As a last resort you can try executing python with `sudo`.
But maybe this is an issue related to Ubuntu 17.10. I never tested beacontools on that OS. Hopefully I have time this weekend to look into it.

williamflynt, 2018-05-27:
I tried adding other capabilities to the `setcap` command with no luck. As I was experimenting, I noticed that I couldn't use `hcitool` directly, but my `lestates` appear to be normal. Any ideas?

[image]

williamflynt, 2018-05-27:
I'll be investigating this post as a possible solution this evening: [StackOverflow Post](https://stackoverflow.com/questions/29099797/raw-capture-capabilities-cap-net-raw-cap-net-admin-not-working-outside-usr-b)

williamflynt, 2018-05-28:
Problem solved. From the above StackOverflow post by Graham King:

> This might be because your home directory is mounted with `nosuid`, which seems to prevent capabilities working. Ubuntu encrypts the home directory, and mounts that with `ecryptfs` as `nosuid`.

This is critical, because capabilities (ie: `setcap`) is what allows this script to function without using `sudo python`. I cloned the project into my `/tmp` directory and there are no issues with running the scan.

closed 2018-05-28 by williamflynt