citruz/beacontools #18: How to make it work

benjilev92, 2018-05-24:
Hi 

Since i want to connect a Cypress CYALKIT-E2 device to a raspberry pi 3, and since Python, Bluetooth and others are configured already, how can i make everything you've done work ?

I know i can find the MAC address easily

citruz, 2018-05-24:
You can use [scanner_cypress_beacon_example.py](https://github.com/citruz/beacontools/blob/master/examples/scanner_cypress_beacon_example.py) as a starting point and replace line 10 with:
```
device_filter=BtAddrFilter(bt_addr="12:3b:6a:1b:4c:65"))
```
Replace with the address of your beacon, of course.
Then execute `python scanner_cypress_beacon_example.py` and you should be seeing data coming from your beacon. If you want to write this data to a file, there are plenty tutorials on how to write to files in Python on the internet.

benjilev92, 2018-05-24:
Hi, 

Thanks for your response,

And what about if i don't have information about the CYALKIT-E02 device and i still want to know its data ? How can i find the CYALKIT-E02 devices and their information with CLI with the rasp, i cannot find them with "bluetoothctl -> scan on" but I can find them from my iPhone with the Beacon Cypress App

regards

citruz, 2018-05-24:
Oh yes, you need to add `BtAddrFilter` to the imports in the second line of the script, sorry.

If you can't see them with `bluetoothctl` there is something wrong. Did you enter `set-scan-filter-clear` before `scan on`? What exactly is the output?

I don't have any Cypress Beacons myself. @darkskiez implemented the support for them in beacontools. Maybe he can help?

benjilev92, 2018-05-25:
Well, done but still doesn't work, and i don't have any filter. There is no output that's the point haha

I am actually trying to take the data from the btmon command and grep, I'll keep in touch on this.

Maybe he can ;)

citruz, 2018-05-25:
Do you see output in btmon? This should include the MAC of the beacon.

benjilev92, 2018-05-25:
Yep i see it.

As I told you i was working on getting the data, i created a bash in order to get information from the command, i'll share it explained here later as a link fo my github but here is the first version if you want 

#!/bin/bash

echo "killing all previous server"
tmux kill-server

echo "Reseting HCI0"
sudo hciconfig hci0 reset

echo "HCI0 down"
sudo hciconfig hci0 down

echo "HCI0 up"
sudo hciconfig hci0 up

echo "creating new session with tmux for bluetoothctl"
tmux new-session -d -s ServerFault 'sudo bluetoothctl -a |& tee /run/shm/BLUETOOTH_OUTPUT'

echo "sending command scan for devices"
tmux send-keys -t ServerFault "scan on" Enter

echo "five second sleep for scanning devices"
sleep 5

echo "sending command info 00:A0:50:29:34:D5"
tmux send-keys -t ServerFault "info 00:A0:50:29:34:D5" Enter

echo "sending command scan off"
tmux send-keys -t ServerFault "scan off" Enter

cat /run/shm/BLUETOOTH_OUTPUT
echo > /run/shm/BLUETOOTH_OUTPUT
echo

darkskiez, 2018-05-28:
The example should work with no modifications for all the beacons, as they ship with the same UUIDs.

I made some rough notes that I meant to expand on but never got around to it so I just published it in rough form anyway:

http://bryars.eu/2018/05/solar-powered-temperature-humidity-ble-beacon-fun-with-linux/

Make sure the switch is on DM and shine your phone torch at it to make it emit the beacons frequently when debugging :)

darkskiez, 2018-05-28:
I assume you also haven't gone into the debug console on the beacon and switched it into eddystone mode or anything odd.

benjilev92, 2018-05-28:
I don't know why it didn't work for me, i didn't look further anyway

I've seen your blog already 

don't worry i was using my phone for the light ;) 

I got everything in control now, check it out here ! 
every thing is explained in my release ;) https://gist.github.com/benjilev92/3eaa790a0244c8bbafed41bec81adf4a

closed 2018-06-01 by citruz