citruz/beacontools #25: Eddystone beacons not listed

dedalozzo, 2018-08-01:
I have 4 Estimote beacons and I have enable Eddystone-UID, Eddystone-EID, Eddystone-URL, Eddystone-TLM broadcasting. They all share the same namespace, that I have used to filter the beacons as per your instruction. Unfortunately the script `scanner_eddystone_example.py` doesn't return any result. No error is returned, the script just goes in timeout.

If I use instead the iBeacon protocol with the script `scanner_ibeacon_example.py` I get all the beacons regularly filtered using their UUID, which is the same for them all.

I'm on a Raspberry Pi with Raspbian Lite and I have updated BlueZ from source to the version 5.50. Python is compiled with Bluetooth support. The command is executed with root privileges.

citruz, 2018-08-02:
Maybe your specific beacons use some identifiers that the library does not recognize yet. Please do the following:
"""
sudo apt-get install bluez-hcidump
sudo hcitool lescan --duplicates &
sudo hcidump --raw
"""
This will dump the raw ble packets to the console. Please post them here so we can have a look at the issue.

citruz, 2018-08-07:
Thanks, I will look into it. What is the exact model you are using?

dedalozzo, 2018-08-07:
It's the **Proximity Beacon**, the first one on the left in the list that you can find at the bottom of the [products](https://estimote.com/products/). The website claims it support both iBeacon and Eddystone simultaneously. But I have tried to enable even just Eddystone without success. There is an app that you can use to configure them, pretty straightforward. You can even set which frames to include in the broadcasting: UID, URL, EID, TLM, etc.

eltoro81, 2018-08-08:
Could it be related to https://github.com/citruz/beacontools/issues/10?

dedalozzo, 2018-08-08:
It could.

citruz, 2018-08-08:
I found the issue, why the Eddystone packets were not captured. In your dump are also some Estimote specific packets. I will add these as a completely new class of packets to the parser. You need to give me a few days, though.

dedalozzo, 2018-08-08:
Sure thing, thank you very much for the prompt response. I’m glad the dump has been helpful.

VPKi, 2018-08-13:
I have somewhat a same issue, I can see Eddystone TLM and URL frames, as well as IBeacon Frames, but Eddystone UID frames elude the scan. How do I know they are broadcasting UID? I have a smartphone app that picks up every bluetooth packet around it, and on it I can see my beacon broadcasting UID frames. Using beacons from kontakt.io

citruz, 2018-08-14:
I just pushed the branch for the next release. It is still work in progress but I already implemented some changes:
* relaxed the parsing constraints (RFU and flags field) -> that should solve @dedalozzo and @eltoro81 's problems.
* added Estimote Telemetry packet (only subframe B at the moment)

With these changes I already see `EddystoneURLFrame`, `IBeaconAdvertisement`, `EddystoneUIDFrame`, `EddystoneTLMFrame` and `EstimoteTelemetryFrameB` packets in the dump @dedalozzo provided.

```
EddystoneURLFrame<tx_power: -22, url: https://twitter.com/estimote>

IBeaconAdvertisement<tx_power: -71, uuid: b9407f30-f5f8-466e-aff9-25556b57fe6d, major: 42782, minor: 64526>

EddystoneUIDFrame<tx_power: -22, namespace: edd1ebeac04e5defa017, instance: 47a038d5eb03>

EddystoneTLMFrame<voltage: 2991 mV, temperature: 26 Celsius, advertising count: 5093979, seconds since boot: 48741845>

EstimoteTelemetryFrameB<identifier: 47a038d5eb032640>
```

citruz, 2018-08-14:
Also added subframe A of the Estimote telemetry packet spec.

dedalozzo, 2018-08-14:
Thank you very much @citruz, can you please make a release so I can get it from pip and test your changes?

dedalozzo, 2018-08-15:
Hi @citruz, I have tested it and it's working properly, thank you:

```
<fc:30:17:20:34:2a, -67> EddystoneTLMFrame<voltage: 3047 mV, temperature: 49178 Celsius, advertising count: 5899888, seconds since boot: 24763922> {'namespace': 'edd1ebeac04e5defa017', 'instance': '5d124ce70ec5'}
<e7:78:c6:dc:67:04, -36> EddystoneTLMFrame<voltage: 2966 mV, temperature: 16409 Celsius, advertising count: 7370726, seconds since boot: 55750462> {'namespace': 'edd1ebeac04e5defa017', 'instance': '47a038d5eb03'}
```

dedalozzo, 2018-08-15:
One last thing. Are you parsing even `VBATT[1]` `0x02` and all the other non standard parts of the TLM frame? I'm trying to find out is some beacon manufacturer is broadcasting the battery life (not just the voltage), or if I have to determine it myself based on some assumptions.

citruz, 2018-08-15:
Yes, `VBATT[0]` and `VBATT[1]` are treated together as a 16 bit integer. I'm not sure what other parts you are referring to. This is the specification I use for parsing:  https://github.com/google/eddystone/blob/master/eddystone-tlm/tlm-plain.md

citruz, 2018-08-15:
No, you can just pass a list as the `packet_filter` argument like this:
```
scanner = BeaconScanner(callback, 
    packet_filter=[EddystoneTLMFrame, EddystoneUIDFrame]
)
``` 
Maybe I should add this to the examples :)

dedalozzo, 2018-08-15:
Good to know. Are you planning to support GATT as well to configure the beacons?

citruz, 2018-08-15:
No sorry.

dedalozzo, 2018-08-15:
Every time the scanner receive a frame, it triggers the callback, so `packet` holds one frame at the time. Since each frame might have a different broadcasting interval, that's the only way I guess.

citruz, 2018-08-15:
You can get the class of a certain packet using `isinstance(packet, EddystoneTLMFrame)`. Or what is your question exactly?

dedalozzo, 2018-08-15:
Not a question, really, I was just describing the way it works. :-)
I guess we can close this. Thank you for the prompt response.

closed 2018-08-15 by dedalozzo

citruz, 2018-08-26:
I just released the new version on PyPi (as 1.3.0 and not as 2.0.0, though).
https://pypi.python.org/pypi/beacontools/