citruz/beacontools #27: iBeacon scanner and raspberry pi 3 b+

eashary, 2018-08-16:
Hi there, 

First, thank you for this great library. 

I use your library in one of my programs on raspberry pi devices to scan for surrounding iBeacon beacons  and report their information. It works perfectly on raspberry pi3 B. However, when I run the scanner_ibeacon.py example on raspberry pi 3 B plus model device, it doesn't return any beacons information. 
Interestingly scanner_ibeacon.py example works and discovers the around beacons after I scan first using bluetoothctl or hcitool utilities.

Thank You

jdhernandez88, 2018-10-08:
I am having the same issue. Same code only works on a Pi 3 B (Stretch) but doesn't work on a Pi 3 B+ (Stretch).  Looking for any suggested solutions.

citruz, 2018-10-22:
Unfortunately, I don't own a Pi 3 B+ and without hands on the hardware I cannot identify the issue. 
Until someone has found a proper solution, you could run a short scan (1s) using hcitool before starting your python scan. For example using this [script](http://federicopfaffendorf.com.ar/hcitool-lescan-timeout/). You could also call this script from within python, before starting the scan.

CoderScrappy, 2018-11-12:
I'm really having problems with the Raspberry Pi 3 model B +. I wish it would be solved as soon as possible. Thanks.

jdhernandez88, 2018-11-12:
As citruz already mentioned, run lescan for about 1 second then run the scan. It works for me as a temporary workaround. If anyone knows a quicker solution/workaround I am open to hear about it.

CoderScrappy, 2018-11-13:
"Leccan" kodunu çalıştırdım, ancak birkaç kez buldum ve sonra olmadı. Döngüde bir veya iki kez buldu.

citruz, 2018-12-04:
#31 contains a fix for the LE scan on the Raspberry Pi 3+. I would be happy for some feedback whether this solves the problem for you, since I cannot test it myself.

RafalSkolasinski, 2018-12-10:
> #31 contains a fix for the LE scan on the Raspberry Pi 3+. I would be happy for some feedback whether this solves the problem for you, since I cannot test it myself.

I am happy to test it. However I am getting some odd errors -> see my comment in the PR.

RafalSkolasinski, 2018-12-16:
This should be fixed by https://github.com/citruz/beacontools/pull/31, no?

ostar0816, 2019-06-21:
This is fixed and should be closed, I tested on my raspberry pi 3 b+. Thank you for the amazing library :)

closed 2020-06-28 by citruz