citruz/beacontools #33: Threading and loops

rkaneknight, 2019-01-09:
I've edited scanner.py to serve my purposes by creating the _mon in the start, and deleting it in the stop.  Not sure if this is a good way to handle it as I don't understand python as well as you do.
I wanted to check the same beacon on a 5 second interval and produce a log of the RSSI values.
But with your code I was trying to start a thread again which yielded an error message.

This is the code change that I made starting at line 51 of scanner.py
        self.callback=callback
        self.bt_device_id=bt_device_id
        self.device_filter=device_filter
        self.packet_filter=packet_filter

    def start(self):
        """Start beacon scanning."""
        self._mon = Monitor(self.callback, self.bt_device_id, self.device_filter, self.packet_filter)
        self._mon.start()

    def stop(self):
        """Stop beacon scanning."""
        self._mon.terminate()
        del(self._mon)

Maybe you have already created a way for me to do what I want, or maybe this is a better approach since the thread is destroyed and recreated on start/stop.

stephane42, 2019-02-14:
you can also start the scan at the begining of you main 
and use **scanner._mon.toggle_scan(True/False**) in your loop to stop scanning during process.

RafalSkolasinski, 2019-03-15:
Not sure if related: when one `start` and then `stop` scanning after trying to `start` it again I am getting:
```
RuntimeError: threads can only be started once
```