citruz/beacontools #34: scanner_eddystone_example.py not working

YoussF, 2019-01-14:
Hello everyone,

I trying to use to example file for scanning EddystoneUID frames... but it looks like it doesn't work. I know that there are similar related closed issues for the same problem (https://github.com/citruz/beacontools/issues/10 and https://github.com/citruz/beacontools/issues/25) and @citruz has replied that this has been fixed in 1.3.0 but it does not seems to be the case (at least for me).

I removed only one line form the original example, the one that is used to filter the namespace.

```
device_filter=EddystoneFilter(namespace="12345678901234678901"),
```

So here is the code i used :

```
import time
from beacontools import BeaconScanner, EddystoneTLMFrame, EddystoneFilter, EddystoneUIDFrame

def callback(bt_addr, rssi, packet, additional_info):
    print("<%s, %d> %s %s" % (bt_addr, rssi, packet, additional_info))

# scan for all TLM frames of beacons
scanner = BeaconScanner(callback, 
    packet_filter=[EddystoneTLMFrame, EddystoneUIDFrame]
)
scanner.start()

time.sleep(10)
scanner.stop()
```

The console is not printing anything, i have several physical EddystoneUID beacons and also some that are emulated with the "Beacon Simulator" android app.

Can somebody please tell me what i am doing wrong ?

Thanks a lot.

ruibrasfernandes, 2019-05-24:
Hi YoussF, have the same problem with a RPI zero w ... even without filtering nothing, UIF frames are not visible, only TLM.
Rui

citruz, 2020-08-25:
Could be the same problem as #39 which was fixed in 2.0.1. Can you please check again?

closed 2020-10-27 by citruz