citruz/beacontools #39: Eddystone UID not seen

404QuotenPole, 2019-03-14:
Hello,
I tryed out the Eddystone example code.
With this I got a few Problems.

When I put in the Namespace of my Beacons, their is 0 output.
When I put URLFrame into filter i got output.
When I put UIDFrame into filter i got 0 output.

So what could be the problem?
With best of wishes,
QuotenPole

citruz, 2019-03-14:
Can you post the exact code you used and the output you get?

404QuotenPole, 2019-03-18:
`import time
import re
from beacontools import BeaconScanner, EddystoneTLMFrame, EddystoneFilter, EddystoneUIDFrame, parse_packet, EddystoneURLFrame

def callback(bt_addr, rssi, packet, additional_info):
     print("<%s, %d> %s %s" % (bt_addr, rssi, packet, additional_info))

scanner = BeaconScanner(callback,
     #device_filter=EddystoneFilter(namespace="namespace"),
     #packet_filter=[EddystoneUIDFrame, EddystoneURLFrame]
)
scanner.start()

time.sleep(5)
scanner.stop()`

That is my code i use, the namespace is my namespace my beacon uses. And i get the ok output for TLM and URL if i put them into the filter, but nothing for UID. And if i put the namespace filter in i see nothing of all of them.

QuotenPole

ruibrasfernandes, 2019-05-24:
Hi, I have the same issue ... can see the Eddystone TLM frames, but not UID frames.
Also, I'm not able to filter TLM Frames based on namespace.
Any news on this?
Regards,
Rui

tetleytealeaf, 2019-08-20:
The parser for EDDYSTONE_UID is off.    The packet that identifies it of a EDDYSTONE_UID type is \xaa\xfe.   Which is there, but it is wrapped inside a packet type of 3.  parser.py is looking for a frame type of SERVICE_DATA_TYPE, which equals 22; not 3.

tetleytealeaf, 2019-08-22:
I looked into it further. At least for me, the problem is that my EDDYSTONE_UID packets have no rfu field.  The missing two bytes cause the parser not to recognize it as an EDDYSTONE_UID packet.   In struct/eddystone.py, the packet structs are defined as:

EddystoneUIDFrame = Struct(
"tx_power" / Int8sl,
"namespace" / Array(10, Byte),
"instance" / Array(6, Byte),
"rfu" / Array(2, Byte)
)

When I commented out the rfu line, beacontools recognizes my EDDYSTONE_UID packets.  But really, it looks like rfu should exist as an optional field.  Commenting out rfu simply defines Eddystone UID to never have rfu.

marttikuutti, 2020-08-11:
Confirm this. All beacons that we use require us to comment out that rfu-field.

citruz, 2020-08-25:
Like you recommended I removed the rfu field since it was not used anyway. Thanks for the help.

commit d213dc03 (2020-08-25): removed rfu field from EddystoneUIDFrame. fixes #39 (#50)

* removed rfu field from EddystoneUIDFrame. fixes #39
closed 2020-08-25 by citruz