citruz/beacontools #40: The filters are rejecting anything

deisi, 2019-06-01:
Hey I try to use this on my arch linux box, but somehow I can't make it work. I try to run:
```python
import time
from beacontools import (BeaconScanner, EddystoneTLMFrame, EddystoneFilter,
                     IBeaconFilter)

def callback(bt_addr, rssi, packet, additional_info):
    print("<%s, %d> %s %s" % (bt_addr, rssi, packet, additional_info))

# scan for all iBeacon advertisements from beacons with the specified uuid
scanner = BeaconScanner(callback,
    #device_filter=IBeaconFilter(uuid="e5b9e3a6-27e2-4c36-a257-7698da5fc140")
)
scanner.start()
```
But this never finds any package at all. So I started looking into details of the library itself. The first problem was:
https://github.com/citruz/beacontools/blob/15a83e9750d0a4393f8a36868e07f6d9458253fe/beacontools/scanner.py#L168-L172

Up to now I have never seen a package passing this check. All I receive fails this check, however I'm at a science conference and there is a crazy amount of beacons around me. I also use the [Android Beacon Simulator](https://play.google.com/store/apps/details?id=net.alea.beaconsimulator&hl=en_US) and stetted up an IBeacon.

As a matter of desperation I removed the check and went on. However maybe not surprisingly, 
https://github.com/citruz/beacontools/blob/15a83e9750d0a4393f8a36868e07f6d9458253fe/beacontools/scanner.py#L177 is now always None. 

Can this be a problem with the Bluetooth stack itself?  Like wrong bluez version or so?

ostar0816, 2019-06-21:
@deisi

citruz, 2019-07-15:
If you don't suppy any `device_filter`, the scanner mode will be set to  `MODE_ALL` by `get_mode`: https://github.com/citruz/beacontools/blob/55b70136e1a13a3c91115cc77cb1844f01f51d94/beacontools/utils.py#L82-L83
So the check should always pass in your case. As you already mentioned, the issue might be in the data supplied by bluez. Do you still have this problem? If yes, a dump of the `pkt` variable would be really useful to debug further.

citruz, 2020-06-28:
Closed due to inactivity.

closed 2020-06-28 by citruz