citruz/beacontools #41: Temperature not reported correctly with EddyStone TLM frames.

chetferry, 2019-06-08:
It looks like there is a signed vs unsigned issue?  I have a beacon inside a cooler with dry ice.  The temperature inside is about -36 degrees C.  My script is reporting 220 degrees C.  If you interpret this number as a two's complement you get the right answer.  Here is what is reported:

<ac:23:3f:2a:e0:4e, -49> EddystoneTLMFrame<voltage: 3216 mV, temperature: 220 Celsius, advertising count: 12513900, seconds since boot: 99146930>

Here is my code:

import time

from beacontools import BeaconScanner, IBeaconFilter

def callback(bt_addr, rssi, packet, additional_info):
    str = "<%s, %d> %s %s" % (bt_addr, rssi, packet, time.time())
    print(str)

# scan for all advertisements from beacons
scanner = BeaconScanner(callback)
scanner.start()
time.sleep(4)
scanner.stop()

Smigg92, 2019-10-27:
This will most likely be due to the beacon you are using. Whilst it might indeed be sending TLM data sometimes certain manufacturers do alter the tlm standard slighty. If you could attach the advatised data for analysis.

myfreeweb, 2020-06-26:
no, the code here is not correct. [android-beacon-scanner](https://github.com/Bridouille/android-beacon-scanner) reads my custom beacon correctly (which just uses [`fixed::types::I8F8.to_bits`](https://docs.rs/fixed/1.0.0/fixed/types/type.I8F8.html) for temperature). beacontools reads "49181 Celsius" and "8219 Celsius" for temps around +30 degrees :D

PR #44 (2020-06-29): Add FreeBSD support, fix various things
- Split BlueZ backend into a separate module (hopefully correctly, please test)
- Add FreeBSD backend
- Fix #41
- Hopefully fix #43 (by not using randomized MAC)
commit ebc8abdd (2020-06-29): Parse Eddystone temperature correctly, fixes #41

- The temperature field is big endian, like the others
- The meaning of fixed point and regular (parsed) temperature fields was swapped around,
  resulting in things like "8129 Celsius" being logged
closed 2020-06-29 by citruz