citruz/beacontools #43: Example Scanner working on Raspberry Pi, but not on Debian Buster Linux Laptop.

aadhi0319, 2020-05-26:
The program works almost perfectly on the Raspberry Pi, but the exact same program does not return any thing (even the devices the Raspberry Pi detects) on my Lenovo Thinkpad X1 Carbon (running Debian Buster). sudo hcitools lescan returns "Set scan parameters failed: Input/output error". Nothing I do from the countless online suggestions seems to make that error go away. I've restarted hci0, restarted my machine, restarted bluetooth services, etc and the error still persists. bluetoothctl however can scan and detect the le iBeacon messages. Can someone help me figure this out?

myfreeweb, 2020-06-26:
Could this be related to the randomized MAC mode?

With my CSR USB dongle, the `LE_Set_Scan_Enable` command is disallowed (`0xc`) (regardless of enable/disable) when random mode is set.

Try changing `address_type=BluetoothAddressType.RANDOM` to `PUBLIC` in `beacontools/scanner.py`.

wingkopu, 2020-08-13:
Hi,

I tried this lib on Ubuntu 20.04 and i got no result just like you said. I try using hcitool to check on the bluetooth message and found that the received packet has different structure as the code expected. If you look in the Monitor.process_packet(scanner.py:168) , you will see that the code validate the ble packet by looking at its Manufacturer byte and Sub-type bytes (4c 00 02 15). Those 4 bytes were expected to be at index 19:23 as you can see in scanner.py(line: 168). 
However, using hcitool i found that the 4 bytes are at index 16:20. After try changing the underlining index, the code start showing the ibeacon received.

I'm not sure if the bluetooth parameter or driver or any setting cause the ble packet be difference. 
I haven't tried the lib on raspberry pi yet. Believe that there are some different between ble on laptop and ble on raspberry pi.

For now, i use index pkt[-26:-22] for ibeacon validation. scanner.py:168 (Monitor.process_packet)

citruz, 2020-08-25:
Hm that sounds interesting. I didn't know that the received data could change based on the OS. Can you upload a raw packet dump (eg. using `sudo hcidump --raw`)?

citruz, 2020-08-29:
I just published 2.0.2 in which I made packet pre-filtering more resilient against slightly different packet layouts. Instead of fixed indices I check whether the packet contains the specific identifier anywhere. Please check if that fixes your problem.

PR #44 (2020-06-29): Add FreeBSD support, fix various things
- Split BlueZ backend into a separate module (hopefully correctly, please test)
- Add FreeBSD backend
- Fix #41
- Hopefully fix #43 (by not using randomized MAC)
closed 2020-10-27 by citruz