citruz/beacontools #46: Input packets after run 'scan on' command with bluetoothctl

peanutstars, 2020-07-03:
Hello

The test computer is Pericision 5530 of DELL and OS is Ubuntu 18.04.

It is not input packets before executing 'scan on' with bluetoothctl program.
Added the debug code as follow in scanner.py.

```
        while self.keep_going:  
            pkt = self.socket.recv(255)  
            hexdump.hexdump(pkt)
```
It shows packets after scan on:

```    
00000000: 04 0E 04 01 05 20 00                              ..... .
00000000: 04 0E 04 01 41 20 00                              ....A .
00000000: 04 0E 04 02 42 20 00                              ....B .
00000000: 04 0F 04 00 02 01 04                              .......
00000000: 04 3E 39 0D 01 10 00 01  33 70 BD FA C5 31 01 00  .>9.....3p...1..
00000010: FF 00 BE 00 00 00 00 00  00 00 00 00 1F 1E FF 06  ................
00000020: 00 01 09 20 02 B4 C7 9E  AF A2 CB 8E EF D6 F8 FA  ... ............
00000030: 1C EC 64 23 6F 6E C7 50  6B 9B 86 54              ..d#on.Pk..T
00000000: 04 3E 39 0D 01 10 00 01  FC 2B CA 07 71 32 01 00  .>9......+..q2..
00000010: FF 00 C0 00 00 00 00 00  00 00 00 00 1F 1E FF 06  ................
00000020: 00 01 09 20 02 8B C2 F1  D7 57 9D 5B 90 34 4A 87  ... .....W.[.4J.
00000030: C5 C4 BD 64 F1 8D 06 39  BA F7 28 AB              ...d...9..(.
00000000: 04 3E 31 0D 01 13 00 01  05 DD 32 2E E7 50 01 00  .>1.......2..P..
00000010: FF 00 B5 00 00 00 00 00  00 00 00 00 17 02 01 06  ................
00000020: 13 FF 4C 00 0C 0E 00 89  31 FF 2F 5A 09 10 F7 8E  ..L.....1./Z....
00000030: 3F 69 AB 04                                       ?i..
```

How can I operation without running 'scan on' command of bluetoothctl ?
Please give me a advice.

citruz, 2020-08-25:
Sorry I don't have the hardware to reproduce/debug the problem.

citruz, 2020-10-27:
This was fixed with #54 in version 2.1.0.