citruz/beacontools #47: Requesting Better Documentation: Get UUID, Major and Minor in python programme as variable?

FalconBFR, 2020-07-12:
Dear all,

Is it possible to grab the UUID of beacons during scanning, save it as a variable, and use it in the python programme? Where do I return the UUID and where do I save it to? This is because I can only see scanner.start()

Are there other functionalities out there or ways that I can grab the UUID,Major,Minor and RSSI values for further use? Many thanks

FalconBFR, 2020-07-12:
Initial Discovery, Please Correct me if I am wrong:
If you want to append the Mac Addresses to an array, declare the array before and outside the callback function and append to the array in the call back function with the variables provided by the demo code as shown below:

My appended code:
`import time
from beacontools import BeaconScanner, IBeaconFilter, IBeaconAdvertisement

all_bt_addr = []
def callback(bt_addr, rssi, packet, additional_info):
    print("<%s, %d> %s %s" % (bt_addr, rssi, packet, additional_info))
    print(bt_addr)
    all_bt_addr.append(bt_addr)
    print(all_bt_addr)

# scan for all iBeacon advertisements from beacons with the specified uuid
scanner = BeaconScanner(callback,
    device_filter=IBeaconFilter(uuid="e5b9e3a6-27e2-4c36-a257-7698da5fc140")
)
scanner.start()
time.sleep(5)
scanner.stop()`

(For IBeacon demo code)

FalconBFR, 2020-07-12:
More Info: If you want to grab the UUID, Major, and minor, they are embedded in the 'additional information' variable. 
'additional information' is a dictionary {uuid: uuidvalue, major: majorvalue, minor: minor value}

citruz, 2020-07-20:
I am not actively working on this library anymore therefore the documentation isn’t as good as it could be. But I see that you figured it out yourself correctly.
Besides the `additional_info` variable, also the `packet` itself is an instance of one if the classes in https://github.com/citruz/beacontools/tree/master/beacontools/packet_types and contains all available data.
Let me know if you have more questions.

closed 2020-07-21 by citruz