citruz/beacontools #52: Error while executing code

guptva5, 2020-09-01:
Please let me know what i am missing. I used exact example code mentioned.

Traceback (most recent call last):
  File "BluetoothSept012020.py", line 4, in <module>
    from beacontools import BeaconScanner, EddystoneTLMFrame, EddystoneFilter#, EddystoneUIDFrame, EddystoneURLFrame
  File "/home/pi/.local/lib/python2.7/site-packages/beacontools/__init__.py", line 3, in <module>
    from .scanner import BeaconScanner
  File "/home/pi/.local/lib/python2.7/site-packages/beacontools/scanner.py", line 17, in <module>
    from .device_filters import BtAddrFilter, DeviceFilter
  File "/home/pi/.local/lib/python2.7/site-packages/beacontools/device_filters.py", line 110
    raise ValueError("bt_addr({}) wasn't a string".format(bt_addr)) from exc
                                                                       ^
SyntaxError: invalid syntax

FalconBFR, 2020-09-01:
You seem to be using python2.7. I think this library is intended for python3.

FalconBFR, 2020-09-01:
You need to install python3 and when running, instead of python file.py, do python3. Also when doing pip package installations, you might have to call pip3 if pip doesn't seem to work. Try googling for more information. I understand, its frustrating :D Hope it helps.

guptva5, 2020-09-01:
Thank you quick response. I tried python 3 also and here is new error....

Error in address
Traceback (most recent call last):
  File "BluetoothSept012020.py", line 20, in <module>
    scanner.start()
NameError: name 'scanner' is not defined

i used pip3 install beacontools

guptva5, 2020-09-01:
So after this error .... i installed pybluez as well using below command. 

python3 -m pip install pybluez

Now i am getting different error

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/home/pi/.local/lib/python3.7/site-packages/beacontools/scanner.py", line 119, in run
    self.set_scan_parameters(**self.scan_parameters)
  File "/home/pi/.local/lib/python3.7/site-packages/beacontools/scanner.py", line 171, in set_scan_parameters
    self.backend.send_cmd(self.socket, OGF_LE_CTL, OCF_LE_SET_SCAN_PARAMETERS, scan_parameter_pkg)
  File "/home/pi/.local/lib/python3.7/site-packages/beacontools/backend/linux.py", line 19, in send_cmd
    return bluez.hci_send_cmd(socket, group_field, command_field, data)
_bluetooth.error: (1, 'Operation not permitted')

Traceback (most recent call last):
  File "BluetoothSept012020.py", line 19, in <module>
    scanner.stop()
  File "/home/pi/.local/lib/python3.7/site-packages/beacontools/scanner.py", line 69, in stop
    self._mon.terminate()
  File "/home/pi/.local/lib/python3.7/site-packages/beacontools/scanner.py", line 259, in terminate
    self.toggle_scan(False)
  File "/home/pi/.local/lib/python3.7/site-packages/beacontools/scanner.py", line 181, in toggle_scan
    self.backend.send_cmd(self.socket, OGF_LE_CTL, OCF_LE_SET_SCAN_ENABLE, command)
  File "/home/pi/.local/lib/python3.7/site-packages/beacontools/backend/linux.py", line 19, in send_cmd
    return bluez.hci_send_cmd(socket, group_field, command_field, data)
_bluetooth.error: (1, 'Operation not permitted')

citruz, 2020-09-16:
Did you execute this command from the readme?
```
# grant the python executable permission to access raw socket data
sudo setcap 'cap_net_raw,cap_net_admin+eip' "$(readlink -f "$(which python3)")"
```