citruz/beacontools #64: ModuleNotFoundError: No module named 'beacontools'

dima72, 2021-03-02:
Hello All,
I've installed beacontools on fresh new RaspberryPi image.
```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)" 

```
sudo apt-get install python3-pip
**pip3 install beacontools**

`pip3 list | grep -i beacontools`   says: **beacontools   2.1.0**

however I run python3, saying `import beacontools`
and get error:
**ModuleNotFoundError: No module named 'beacontools'**

what can be wrong?

iSayeed, 2022-11-10:
try pip install rather than pip3?

dima72, 2022-11-10:
Appreciated!
  
>Четверг, 10 ноября 2022, 7:45 -05:00 от Anas ***@***.***>:
>try pip install rather than pip3?