citruz/beacontools #65: No scanning possible on Raspberry 4B

ergo70, 2021-03-23:
I tried the scanner examples for iBeacons and Eddystone Beacons on a Raspberry PI Zero W and they work fine. When doing the same on a Raspberry 4B the examples work in the sense that no errors are thrown, but they fail in finding any Beacons.

OS is Raspbian Buster
beacontools[scan] is 2.1.0
bluez is 5.50
Python 3.7.3

Any advice how to debug this?

ergo70, 2021-03-23:
I could fix it with #63, setting hci_version to BT_CORE_SPEC_4_2. However, why does this work on the zero when it is somehow related to Bluez, since both run the same Version?

shaojim12, 2021-05-08:
@ergo70 Do you solve the problem of letting Rpi 4 scanning the beacon?
May I ask you how did you solve this problem?
Cause I have a similar problem with this.
Thanks a lot!

mohr-michael-adam, 2021-06-01:
> I could fix it with #63, setting hci_version to BT_CORE_SPEC_4_2. However, why does this work on the zero when it is somehow related to Bluez, since both run the same Version?

https://www.raspberrypi.org/products/raspberry-pi-zero-w/
https://www.raspberrypi.org/documentation/faqs/#:~:text=The%20Raspberry%20Pi%204%20Model,wireless%20LAN%2C%20and%20Bluetooth%205.0.

From what I'm seeing the in the links above the zero w has Bluetooth 4.1 while 4 has Bluetooth 5.

mrts, 2021-06-12:
> @ergo70 Do you solve the problem of letting Rpi 4 scanning the beacon?
> May I ask you how did you solve this problem?
> Cause I have a similar problem with this.

Hi! We were able to receive `ExposureNotificationFrame`s on a Raspberry Pi 4 B with the following code:

```py
import signal
from beacontools.scanner import Monitor, HCIVersion

def callback(bt_addr, rssi, packet, additional_info):
    print("<%s, %d> %s %s" % (bt_addr, rssi, packet, additional_info))

monitor = Monitor(callback,
                  bt_device_id = 0,
                  device_filter = None,
                  packet_filter = None,
                  scan_parameters = {})
monitor.get_hci_version = lambda: HCIVersion.BT_CORE_SPEC_4_2
monitor.start()
signal.pause()
```

However, sadly, Eddystone frames did not come through.

Akkiesoft, 2022-01-14:
I have same issue. And I seem #71 and #73 are same problem.

My workaround was use the previous version to avoid the extended ble command support in the latest version:

```
sudo pip3 uninstall beacontools
sudo pip3 install beacontools[scan]==2.0.2
```

citruz, 2022-02-05:
Unfortunately, I don't have a PI4 to debug this. Could you send me what `get_hci_version` returns?
```
from beacontools.scanner import Monitor

def callback(bt_addr, rssi, packet, additional_info):
    print("<%s, %d> %s %s" % (bt_addr, rssi, packet, additional_info))

monitor = Monitor(callback,
                  bt_device_id = 0,
                  device_filter = None,
                  packet_filter = None,
                  scan_parameters = {})
print(monitor.get_hci_version())
```

Akkiesoft, 2022-02-05:
I ran that code but I got ```Segmentation fault```.

When I use ```print(monitor.hci_version)``` instead of ```print(monitor.get_hci_version())```, it returns ```HCIVersion.BT_CORE_SPEC_1_0```.

citruz, 2022-02-06:
Oh yeah, my bad. The `get_hci_version` requires the socket to be already opened, which is done in `monitor.start()`. Executing `start()` before `get_hci_version()` should yield a proper result.

Akkiesoft, 2022-02-07:
Thank you for your advice :)

I got below result:

```
3B : HCIVersion.BT_CORE_SPEC_4_1
3B+: HCIVersion.BT_CORE_SPEC_5_0
4B : HCIVersion.BT_CORE_SPEC_5_0
```

drdas, 2022-12-14:
> > @ergo70 Do you solve the problem of letting Rpi 4 scanning the beacon?
> > May I ask you how did you solve this problem?
> > Cause I have a similar problem with this.
> 
> Hi @shaojim12! I was able to receive `ExposureNotificationFrame`s on a Raspberry Pi 4 B with the following code:
> 
> ```python
> import signal
> from beacontools.scanner import Monitor, HCIVersion
> 
> def callback(bt_addr, rssi, packet, additional_info):
>     print("<%s, %d> %s %s" % (bt_addr, rssi, packet, additional_info))
> 
> monitor = Monitor(callback,
>                   bt_device_id = 0,
>                   device_filter = None,
>                   packet_filter = None,
>                   scan_parameters = {})
> monitor.get_hci_version = lambda: HCIVersion.BT_CORE_SPEC_4_2
> monitor.start()
> signal.pause()
> ```
> 
> However, sadly, Eddystone frames did not come through, even though [node-beacon-scanner](https://github.com/futomi/node-beacon-scanner) was able to receive them on the same device.
> 
> @citruz, do you have any ideas how to proceed?

Same exact observation on Nvidia Jetson TX2 with JetPack 4.6.2 - it works with hcitool, bluetoothctl and other tools. While the above snippet works to scan for iBeacons (and other BLE devices), the beacon scanner example in beacontools produces no output at all. Any ideas as to how to fix this properly?

drdas, 2022-12-14:
> > > @ergo70 Do you solve the problem of letting Rpi 4 scanning the beacon?
> > > May I ask you how did you solve this problem?
> > > Cause I have a similar problem with this.
> > 
> > 
> > Hi @shaojim12! I was able to receive `ExposureNotificationFrame`s on a Raspberry Pi 4 B with the following code:
> > ```python
> > import signal
> > from beacontools.scanner import Monitor, HCIVersion
> > 
> > def callback(bt_addr, rssi, packet, additional_info):
> >     print("<%s, %d> %s %s" % (bt_addr, rssi, packet, additional_info))
> > 
> > monitor = Monitor(callback,
> >                   bt_device_id = 0,
> >                   device_filter = None,
> >                   packet_filter = None,
> >                   scan_parameters = {})
> > monitor.get_hci_version = lambda: HCIVersion.BT_CORE_SPEC_4_2
> > monitor.start()
> > signal.pause()
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > However, sadly, Eddystone frames did not come through, even though [node-beacon-scanner](https://github.com/futomi/node-beacon-scanner) was able to receive them on the same device.
> > @citruz, do you have any ideas how to proceed?
> 
> Same exact observation on Nvidia Jetson TX2 with JetPack 4.6.2 - ble scanning works with hcitool, bluetoothctl and other tools. While the above snippet works to scan for iBeacons (and other BLE devices), the beacon scanner example in beacontools produces no output at all. Any ideas as to how to fix this properly?

UPDATE: The suggestion above from @Akkiesoft worked to fix the problem with the older version of beacontools!

agentgonzo, 2024-03-22:
I'm facing a similar situation to the above, except that I'm using a raspberry pi zero and a pi zero 2.
Both have the same setup:
* beacontools 2.1.0
* bluez 5.53

I have the same scanning code as above (without modifying the hci_version).
If I run this on the pi zero, it gets beacon data every second (every time the beacon emits a pulse).
If I run this on the pi zero 2, it gets no data.

However, if I run the following commands on the pi before running the python script:
```
bluetoothctl
scan on
quit
```
Then beacontools starts logging data... but not all the time. It seems to log it for "a while" (a few minutes or so), then stop, then restart.  I have no idea why the behaviour is different, or why interacting with bluetoothctl seems to kick something into life