citruz/beacontools #67: Error on Raspberry Pi OS - _bluetooth.error: (9, 'Bad file descriptor')

Andperro, 2021-04-30:
Hi,

I'm facing a problem with Bluez library when running Raspberry Pi OS on a Raspberry Pi Zero W.

I'm running the latest image version with release date in March 4th 2021.

I used the example on the pypi.org page, as shown below.

```python
import time
from beacontools import BeaconScanner, IBeaconFilter

def callback(bt_addr, rssi, packet, additional_info):
    print("<%s, %d> %s %s" % (bt_addr, rssi, packet, additional_info))

# scan for all iBeacon advertisements from beacons with the specified uuid
scanner = BeaconScanner(callback,
    device_filter=IBeaconFilter(uuid="e5b9e3a6-27e2-4c36-a257-7698da5fc140")
)
scanner.start()
time.sleep(5)
scanner.stop()

# scan for all iBeacon advertisements regardless from which beacon
scanner = BeaconScanner(callback,
    packet_filter=IBeaconAdvertisement
)
scanner.start()
time.sleep(5)
scanner.stop()

```

I get always the same error, as described below.

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/home/pi/.local/lib/python3.7/site-packages/beacontools/scanner.py", line 144, in run
    self.socket = self.backend.open_dev(self.bt_device_id)
  File "/home/pi/.local/lib/python3.7/site-packages/beacontools/backend/linux.py", line 13, in open_dev
    socket.setsockopt(bluez.SOL_HCI, bluez.HCI_FILTER, filtr)
_bluetooth.error: (9, 'Bad file descriptor')
```

linjmeyer, 2022-01-30:
@Andperro did you ever get anywhere with this?  I have the same issue with Raspberry PI OS and beacontools

madfelps, 2022-01-31:
Hello everyone, any update about this error? I'm facing the same issue, and apparently my bluetooth device is available because when I type $ hcitool dev it shows the hci0 address

linjmeyer, 2022-02-13:
I have this issue with an Ubuntu laptop and Raspberry PI (Raspberry OS), with 2 different bluetooth adapters as well.  What I am running is a background daemon and the beacons stop coming in after this error shows in the logs.  I wonder if there is a way to catch or detect this so I can restart the scanner?  Restarting the daemon works everytime so it would be great to code that in.