citruz/beacontools #73: not receiving any data

bbakerzzz, 2022-01-05:
Hello and thanks for your work

I am trying to build a Python scripts to collect Temp and Humidity from some beacons.

I downloaded your code to a Raspberry pi 4 and I do not receive any information.  Should this could work on a RPI4?

When I use  bluetoothctl I see two devices.  Below is a sample of what I see.

[CHG] Device DD:34:02:06:D0:60 ServiceData Key: 0000feaa-0000-1000-8000-00805f9b34fb
[CHG] Device DD:34:02:06:D0:60 ServiceData Value:
  21 01 0f 0e 07 1a 5f 1f 7e 00 11 ff fc 04 06     !....._.~......
[CHG] Device DD:34:02:06:D0:60 ServiceData Key: 00002080-0000-1000-8000-00805f9b34fb
[CHG] Device DD:34:02:06:D0:60 ServiceData Value:
  6d 01 00 00 00 00                                m.....
[CHG] Device D0:AA:E6:17:56:99 ServiceData Key: 00000318-0000-1000-8000-00805f9b34fb
[CHG] Device D0:AA:E6:17:56:99 ServiceData Value:
  24 2b 15 2f 00 00 00 00 00 00 00 00 00 00 00 00  $+./............
[CHG] Device D0:AA:E6:17:56:99 ServiceData Key: 00001803-0000-1000-8000-00805f9b34fb
[CHG] Device D0:AA:E6:17:56:99 ServiceData Value:
  d0 aa e6 17 56 99 00 01 00 02 06 03 e8 64        ....V........d

Thanks in advance for your help.

drdas, 2022-12-14:
The new BLE functionality in version 2.1.X breaks on Jetson TX2 and many other embedded devices. Please stick to version 2.0.2 until @citruz proposes a backward compatibility fix.

Ming119, 2023-11-22:
> The new BLE functionality in version 2.1.X breaks on Jetson TX2 and many other embedded devices. Please stick to version 2.0.2 until @citruz proposes a backward compatibility fix.

I love you bro, you solve my problem haha